AlexanderMandera/arduino-wch32v003 #6: Adding board to Arduino IDE

phpbbireland, 2023-06-06:
Just to mention...
While trying to add board, it reports a CRC mismatch, retry later...

AlexanderMandera, 2023-06-07:
Can you explain which error is shown while adding the board?
I have checked the SHA256 checksum and it should be fine...

phpbbireland, 2023-06-07:
Board Manager reports: CRC doesn't match, file is corrupt. It may be a temporary problem, please retry later.
Console: ```CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:101)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:56)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:108)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more```

phpbbireland, 2023-06-07:
Just installed without issues under Windows 11... 
An OS issue?

AlexanderMandera, 2023-06-07:
I see, thank you for the details.

It was indeed a problem with the checksums, but not with the board/core itself
but with the `minichlink` binary archives that I use to flash the chip.

It looks like I have accidentally updated the checksum only for the Windows version but not for Linux or macOS.

The change is applied (https://github.com/AlexanderMandera/arduino-wch32v003/commit/738c9a94fe3ecb6bdb441dfdfd6cf5defe4ccc14) and should work now.
I have successfully tested the checksum for macOS but not for Linux yet.

phpbbireland, 2023-06-07:
Perfect, install worked fine... many thanks.

Mike

closed 2023-06-07 by phpbbireland