AlexanderMandera/arduino-wch32v003 #9: Error: couldn't open wch link device

wlkmanist, 2023-07-18:
Uploading using MounRiver works fine.

[image]

AlexanderMandera, 2023-07-18:
"In Windows, if you want to use minichlink with the LinkE, you will need to use [Zadig](https://zadig.akeo.ie/) to install WinUSB to the WCH-Link interface 0." (ch32v003fun wiki)

This is currently a limitation of minichlink that it doesn't work with the WCH USB drivers. On Linux and Mac, you don't need that. Maybe at some point, this won't be needed but currently it is.

If you want to use WCH software to program again, you can uninstall the WCH LinkE device in device manager and reconnect it. Then it gets the old drivers again.

This is something I have missed to document, which is only documented in ch32v003fun. I will add that to the README soon.

PTSolns, 2023-08-06:
I'm having the same issue as OP. Tried using the Link-E programmer, as well as others such as Sparkfun Serial Basic board, and others USB-to-UART. Nothing worked. I'm getting same errors as above image shows. Can you please provide more detailed steps as to what needs to be done to resolve this.. The above instructions are not clear. Thanks!

AlexanderMandera, 2023-08-06:
@PTSolns I have summarized the additional steps required for the Link-E programmer to work with `minichlink`:
https://github.com/AlexanderMandera/arduino-wch32v003/wiki/Additional-Installation-Steps
Let me know if there are any problems with this.

USB-to-UART programmers don't work as V003 use the SWIO protocol for programming.

More programmers (mostly DIY options) can be found here:
https://github.com/cnlohr/ch32v003fun/wiki/Features#minichlink

You can also use an Arduino Uno or similar AVR boards can be used as SWIO programmers as well and should be supported by `minichlink`: https://gitlab.com/BlueSyncLine/arduino-ch32v003-swio

PTSolns, 2023-08-06:
@AlexanderMandera Thanks for the speedy reply! I did the above steps and now am getting closer. I have a different set of error messages appear in Arduino IDE. The sketch I am trying to run is as follows.

```
uint32_t count;

void setup() {
  pinMode(D0, OUTPUT);
  pinMode(C0, OUTPUT);
}

void loop() {
	digitalWrite(D0, LOW);
	delay(1000);
	digitalWrite(D0, HIGH);
	delay(2000);
  digitalWrite(C0, LOW);
	delay(100);
	digitalWrite(C0, HIGH);
	delay(100);
	count++;
}
```

The resulting error messages are:
[image]

Below is my physical hardware connection between the Link-E and the nanoCh32V003 v1.0

Link-E -> nanoCH32V003
U_RX -> D5
U_TX -> D6
GND -> GND
3V3 -> 3V3

Any thoughts? Thanks again.

AlexanderMandera, 2023-08-06:
You need to connect the SWIO pin of the WCH-LinkE to D1, 3V3 to 3V3 and GND to GND, no UART pins are necessary for flashing

PTSolns, 2023-08-06:
@AlexanderMandera great I now got it working. Am blinking the onboard LED on D6. Thanks a bunch for the walk-through help.

One thing I am unclear about is why the SWIO pin must be used. Is there no way to use UART pins? I'd like to be able to use other programmers and not the Link-E. For example, I like using the CH340C for UART. Is there a way to get that to work here?

AlexanderMandera, 2023-08-06:
For the CH32V003 there is no UART bootloader available, that's why the debug interface has to be used.

You could write and flash your own bootloader for that, but that's not trivial and you always need to flash the bootloader the first time using SWIO.

Other programmers like an ESP32-S2 or Arduino Uno-based solution where they implemented the SWIO protocol themselves, are listed in the two links in my previous post.

micropythondevel, 2023-08-12:
riscv-none-elf-g++: warning: '-x c++' after last input file has no effect
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
Failed uploading: uploading error: exit status 224

help please is linux ....

AlexanderMandera, 2023-08-12:
Try to follow these instructions:
https://github.com/cnlohr/ch32v003fun/wiki/Installation#non-root-access-on-linux

The udev rules file is also found in the ch32v003fun repository.

I will try to add this to the pre-installation script soon if possible.

closed 2023-08-15 by AlexanderMandera