AlexanderMandera/arduino-wch32v003 #13: No analogread :/

MangyCat, 2023-11-12:
I tried compiling an sketch for my ch32v003, everything else works fine, except analogwrite aswell and analogread i got this error

riscv-none-elf-g++: warning: '-x c++' after last input file has no effect
/home/etore/.arduino15/packages/alexandermandera/tools/riscv-none-elf-gcc/12.2.0-3-xPack/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld: /tmp/ccBBIZmj.ltrans0.ltrans.o: in function `.L0 ':
/home/etore/Arduino/pcmplayerch32v003/pcmplayerch32v003.ino:9: undefined reference to `analogRead' (this is the important part)
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board CH32V003.

And also an undefined reference to "analogwrite"

I understand you haven't implemented everything yet, but i would love for analogwrite and analogread to be implemented instead of having to resort to using digitalwrite and simulated pwm instead of analogwrite

pixelEDI, 2023-11-12:
Hey, so far there is no analogRead implemented. I'm also waiting hard for this :)

For analogWrite you can try something like this

digitalWrite(ledPin, HIGH);
  delayMicroseconds(pwmValue);
  digitalWrite(ledPin, LOW);
  delayMicroseconds(255 - pwmValue);

AlexanderMandera, 2025-11-30:
> > Judging by the time, it will not be implemented
> 
> [image]

I want to work on a new version during the Christmas holidays, using the new ch32fun version and with support for all supported chips.

You can always use examples from ch32fun as a reference for analog read if it is not implemented.