AlexanderMandera/arduino-wch32v003 #15: Does not support Ardulink as a programmer "miniblink -c /dev/ttyACM0"

zoobab, 2023-12-28:
Hi,

I am succesfully using Ardulink as a programmer (with a 1K resistor) to flash the ch32v003:

https://gitlab.com/BlueSyncLine/arduino-ch32v003-swio

Could you add this programmer as a option, I just needed to add "-c /dev/ttyACM0" as an option to make it work.

zoobab, 2023-12-28:
When I try to add the option -c /dev/ttyACM0 in the platform.txt (maybe not the best place), I have the impression that minichlink 1.0.0 is too old and does not support Ardulink. When I try with the most recent git version of minichlink it works fine. How do you define this version 1.0.0?

AlexanderMandera, 2023-12-28:
Hi,
the minichlink binaries are distributed through another GitHub Repo (https://github.com/AlexanderMandera/minichlink-binaries).
It is like that because ch32v003fun only provides Windows x64 binaries and other platforms need to build from source.

The last release build was in June, I haven't updated to the newest version yet
and haven't got any automation yet that supports all platforms at once.

I will update the minichlink builds soon and add more programmer options in the menu.
In the mean time, you can replace the binary in `packages/alexandermandera/tools/minichlink/1.0.0` which should fix your issue.

zoobab, 2024-04-12:
"haven't got any automation yet that supports all platforms at once."

I could try to work on that.

If you want I can try to build more recent binaries as well.

AlexanderMandera, 2024-04-12:
Hi, the problem is supporting the ARM architectures, and especially MacOS.
I think there is something in the making to get ARM based Linux runners, but it seems to be in private beta?

zoobab, 2024-04-12:
On Fri, Apr 12, 2024 at 2:59 PM Alexander Mandera ***@***.***>
wrote:

> Hi, the problem is supporting the ARM architectures, and especially MacOS.
> I think there is something in the making to get ARM based Linux runners,
> but it seems to be in private beta?
>

On github actions, maybe. You could attach a slave ARM server via Scaleway,
I could sponsor one.

Or spawn some qemu arm VM inside an x64 slave.

-- 
Benjamin Henrion (zoobab)
Email: zoobab at gmail.com
Mobile: +32-484-566109
Web: http://www.zoobab.com
FFII.org Brussels
"In July 2005, after several failed attempts to legalise software patents
in Europe, the patent establishment changed its strategy. Instead of
explicitly seeking to sanction the patentability of software, they are now
seeking to create a central European patent court, which would establish
and enforce patentability rules in their favor, without any possibility of
correction by competing courts or democratically elected legislators."