AlexanderMandera/arduino-wch32v003 #19: unable to use A2 as GPIO output

chaealarm, 2024-04-08:
`#define ledpin A2
#define ledpin2 A1

void setup() {
  pinMode(ledpin, OUTPUT);
  pinMode(ledpin2, OUTPUT);
}

void loop() {
  digitalWrite(ledpin, HIGH);  
  delay(1000);   
  digitalWrite(ledpin, LOW);   
  delay(1000);        
  digitalWrite(ledpin2, HIGH);   
  delay(1000);                       
  digitalWrite(ledpin2, LOW);    
  delay(1000); 
}`

At this code, A1 is blinking, but A2 is floating.
These pin was no problem when using ch32v003fun.

I think it might be a compiler problem(ch32v003fun : riscv62-unkonwn-elf, arduino-wch32v003 : riscv-none-elf)

rsazenha, 2024-04-08:
#20 this commit fix your issue. Check modification on file wiring_digital.cpp and change your file. 

Line 52 change from this:

`static uint8_t gpioPin(uint8_t gpio, pin_size_t pin)`
`{`
`    if(gpio == 0) {`
`        return pin;`
`    } else if(gpio == 2) {`
`        return pin - 2;`
`    } else {`
`        return pin - 10;`
`    }`
`}`

To this:

`static uint8_t gpioPin(uint8_t gpio, pin_size_t pin)`
`{`
`    if(gpio == 0) {`
`        return pin + 1;`
`    } else if(gpio == 2) {`
`        return pin - 2;`
`    } else {`
`        return pin - 10;`
`    }`
`}`

chaealarm, 2024-04-09:
> #20 this commit fix your issue. Check modification on file wiring_digital.cpp and change your file.
>
> Line 52 change from this:
>
> `static uint8_t gpioPin(uint8_t gpio, pin_size_t pin)` `{` ` if(gpio [...]

It works! Thank you very much!