AlexanderMandera/arduino-wch32v003 #24: Question - Does this support arduino's tone() function?

alx-alexpark, 2024-08-22:
Hello!

I am looking to use this chip to beep a melody on a passive piezoelectric buzzer.

I plan to use arduino's builtin `tone();` function, is this supported with this microcontroller? Will it only work on certain pins?

Thank you! -- also this is a very cool microcontroller, excited to use!

AlexanderMandera, 2024-08-23:
Hi,
this Arduino Core is currently not up-to-date, I need to find some more time for that. So tone isn't implemented yet.

I plan to rewrite it in the future and add more features and different CH32V chips, but that will take a while.

You can reuse some PWM code like this example to re-create it:
https://github.com/cnlohr/ch32v003fun/blob/master/examples/tim1_pwm/tim1_pwm.c

But there is no easy way that is built-in right now.

I recommend to use ch32v003fun for now if you can (if you want, there is also PlatformIO support).
This core is based on that project and currently has many examples to work with.

pixelEDI, 2024-08-23:
Hey,

A workaround is to use the TimerFreeTone Library: https://bitbucket.org/teckel12/arduino-timer-free-tone/wiki/Home

I made a few PCBs with the CH32V003J4M6 and piezos, and they work pretty well with this Core.

Cheers,  
Edi

closed 2025-01-06 by alx-alexpark